jlecomte/ascom-oag-focuser #1: EEPROM.h: No such file or directory

OriomMao, 2022-05-28:
I use Seeeduino XIAO，upload the code，it show：
`Arduino:1.8.19 (Windows Store 1.8.57.0) (Windows 10), 开发板："Seeeduino XIAO, Arduino, Off, USART"

Arduino_Firmware:11:10: fatal error: EEPROM.h: No such file or directory

compilation terminated.

exit status 1

EEPROM.h: No such file or directory

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。
`
Then I use Arduino uno， it‘s OK

jlecomte, 2022-05-28:
Interesting! Indeed, it seems like the Seeeduino Xiao does not have any EEPROM. I have a few units myself but had not yet tried on them (so far, I've been developing the firmware using an Uno clone) But it seems like there are several options:

1. Option #1 is to not use EEPROM. All the EEPROM-related code would be in `#ifdef ... #endif` blocks... This is not so great because you lose the ability to remember the position of the focuser during a power cycle, which is kind of important.
2. Option #2 is to use something like this: https://github.com/khoih-prog/FlashStorage_SAMD. Unfortunately, this also has a downside. The number of writes will be limited to about 10,000 times. Once you get in that neighborhood, your chip is dead.
3. Option #3 is to use a separate EEPROM chip and use that instead of relying on the micro-controller.
4. Option #4 is to not use a Xiao. I love that little microcontroller board, but for this application, maybe it's not ideal. I am going to experiment with an ATTiny85 board to see if it works well. I will update the README accordingly.

Any other idea @OriomMao ?

jlecomte, 2022-05-29:
I updated the README to warn people of this compatibility issue and to let them know what to do. See https://github.com/jlecomte/ascom-oag-focuser/#microcontroller-compatibility

closed 2022-05-29 by jlecomte