jlecomte/ascom-oag-focuser #2: ASCOM.DarkSkyGeek.OAGFocusDriver.dll

JoeBleau1, 2022-09-27:
Hello, I'm not able to register to my windows 10 pc the ASCOM.DarkSkyGeek.OAGFocusDriver.dll.
I follow the steps indicated but always end with the same message. I even created the dll with visual studio whitout any troubles, but I try to register the dll and those messages appear:

RegAsm : warning RA0000 : Registering an unsigned assembly with /codebase can cause your assembly to interfere with other applications that may be installed on the same computer. The /codebase switch is intended to be used only with signed assemblies. Please give your assembly a strong name and re-register it.
RegAsm : error RA0000 : An error occurred inside the user defined Register/Unregister functions: System.TypeInitializationException: The type initializer for 'ASCOM.DarkSkyGeek.FilterWheelProxy' threw an exception. ---> System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseDecimal(String value, NumberStyles options, NumberFormatInfo numfmt)
   at System.Convert.ToDecimal(String value)
   at ASCOM.DarkSkyGeek.FilterWheelProxy..cctor()
   --- End of inner exception stack trace ---
   at ASCOM.DarkSkyGeek.FilterWheelProxy.RegUnregASCOM(Boolean bRegister)

 I looked at the RA0000 error on the net but this went in a spiral of wouldbe solutions...

Is there a simple solution to this issue ?

Best regards

jlecomte, 2022-09-27:
See **step 2** in https://github.com/jlecomte/ascom-oag-focuser#downloading-and-installing-the-driver. Are you running the command prompt (`cmd.exe`) **as an administrator**?

JoeBleau1, 2022-09-27:
Yes I run the cmd.exe as administrator.

jlecomte, 2022-09-27:
ok, I know what's causing it! It's this line:
https://github.com/jlecomte/ascom-oag-focuser/blob/master/ASCOM_Driver/FilterWheelProxyDriver.cs#L76

Alright, I'll fix it tonight and publish a new release. Thank you for letting me know!

JoeBleau1, 2022-09-27:
You are so cool !
Thank's

jlecomte, 2022-09-28:
OK, please try using the files published in release `v1.0.2`: https://github.com/jlecomte/ascom-oag-focuser/releases/tag/v1.0.2

JoeBleau1, 2022-09-28:
Just did. Still not working, here is the output messages.
**************
RegAsm : warning RA0000 : Registering an unsigned assembly with /codebase can cause your assembly to interfere with other applications that may be installed on the same computer. The /codebase switch is intended to be used only with signed assemblies. Please give your assembly a strong name and re-register it.
RegAsm : error RA0000 : An error occurred inside the user defined Register/Unregister functions: System.TypeInitializationException: The type initializer for 'ASCOM.DarkSkyGeek.FilterWheelProxy' threw an exception. ---> System.FormatException: Input string was not in a correct format.
   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
   at System.Number.ParseDecimal(String value, NumberStyles options, NumberFormatInfo numfmt)
   at System.Convert.ToDecimal(String value)
   at ASCOM.DarkSkyGeek.FilterWheelProxy..cctor()
   --- End of inner exception stack trace ---
   at ASCOM.DarkSkyGeek.FilterWheelProxy.RegUnregASCOM(Boolean bRegister)

jlecomte, 2022-09-28:
OK, one last try, if this one does not work, then I have no idea: https://github.com/jlecomte/ascom-oag-focuser/releases/tag/v1.0.3. Let me know! Thanks.

JoeBleau1, 2022-09-28:
Julien,
Bravo ! It worked, the dll is now registered. 
I can now connect the app, and go to settings etc...
Thank you very much.

JoeBleau1, 2022-09-28:
Have a great evening.
Best regards.

closed 2022-09-28 by JoeBleau1

jlecomte, 2022-09-28:
Brilliant! Thanks for the bug report :)

JoeBleau1, 2022-09-28:
Julien you are more then welcome. :)

ASCOM.DarkSkyGeek.OAGFocusDriver.dll (Issue #2)

 

<https://github.com/jlecomte/ascom-oag-focuser/issues/2#issuecomment-1260386
PFRNANCNFSM6AAAAAAQXFVNSM> .
<https://github.com/notifications/beacon/A3J2AUGJQKB2JVGSDDUTPXLWAPFRNA5CNFS
M6AAAAAAQXFVNSOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTS