jlecomte/ascom-oag-focuser #3: Do you think I can be successful?

Wellco, 2023-02-21:
Hi Julien, I would love to give this project a go, but I have a few concerns mostly around my ability. I will need to modify the 3D files as I have an Antia focuser, OAG and filter draw, not the ZWO. I have zero skills in FreeCAD, but have decided to use Fusion 360 as I think that might be easier. I have already made some basic edits. The Antila is a touch bigger than the ZWO. I don't have a 3D printer, but will get the parts professionally printed. My next issue is the electronics where my knowledge is at an absolute zero. I would have to follow the information in here on GitHub verbatim. There would be no way for me to stray form the path. If I could not get the exact same components I would most likely be lost. However I would really like to see how far I can get. Would you be willing to help me if I get stuck?

Lastly should I hold on for the revised focuser with that battery, of should I crack on with the original design, as documented here on GitHub.

Thanks so much for sharing your wonderful project and knowledge here.

Tim CORSO

jlecomte, 2023-02-21:
Hi Tim,

Let me summarize:

1) Your equipment differs from the one for which this project was originally designed for.
2) You don't have much (any?) experience with 3D modeling.
3) You don't own a 3D printer.
4) You don't know anything about electronics, nor have the equipment needed to complete an electronics project.

So, to your question "will I be successful?", the answer is an unequivocal no, unfortunately.

To your other question "will I support you?", the answer is also no, unfortunately. I am willing to fix bugs or implement enhancements or answer questions, but I simply do not have the time to provide the full support that you would likely need to make progress on your project.

I hope this does not come out too harsh, but this answer will likely save you a bunch of time and frustration.

Maybe look at Pegasus Astro SCOPS OAG?

closed 2023-02-21 by jlecomte

Wellco, 2023-02-21:
Hi Julien,

I appreciate your frank reply.

Points 1,2,3 are not really a problem, I worked for a CAD company most of my working life, just don't know FreeCAD or Fusion 360. The only part I would need to ask questions around would be the electronics and maybe some aspects of the software. I realise that you could answer a mountain of basic questions, and I wouldn't except that. May be there is more complexity to your project than meets my eye.

Pegasus solution is nice, but way to pricey for me at thus time. I may still have a go and see where I can get to. I may fail, but it will keep me amused while the clouds are paying a visit.

Thanks again

Tim