jlecomte/ascom-oag-focuser #6: unhandled exception in focuser app

CheekyMonkey, 2023-12-19:
Hi Julien

There is a condition in the focuser app that will give an unhandled exception

The maximum target position of the focuser is 4000
Entering a number for the target position greater than 400 will give an unhandled exception.

See below:

[image]

By the way, cool project. I made a focuser, but I made a no-solder version :)

[image]

jlecomte, 2023-12-19:
This is 100% normal and expected. If you want to go beyond 4000 steps, you have to update the maximum position in the driver settings dialog. 4000 is just the default value... I also recommend setting the zero position first. See https://www.youtube.com/watch?v=zPPIgVuxk4Y for more details.

closed 2023-12-19 by jlecomte

CheekyMonkey, 2023-12-19:
> This is 100% normal and expected. If you want to go beyond 4000 steps, you have to update the maximum position in the driver settings dialog. 4000 is just the default value... I also recommend setting the zero position first. See https://www.youtube.com/watch?v=zPPIgVuxk4Y for more details.

Ah, thanks.
I need to read the manual :)