jlecomte/ascom-oag-focuser #9: Confused about hole sizes in base and cover

Endymion42, 2024-02-08:
I'm confused about the hole sizes in the base and cover.  The base holes are significantly smaller than the 4 mount holes in the cover.  As the bolt/screw needs to go through the base into the cover, how is this supposed to work?  

Thanks for any help.  In the mean time I will use a feature in my slicer to fill in the cover holes and replace with smaller diameter holes.

jlecomte, 2024-02-09:
That is because the large holes are designed to host brass inserts. Please see the BOM. Thanks!

Endymion42, 2024-02-09:
Thank you for resolving my confusion.   Apologies, I looked around for a BOM but didn't see one listing the brass inserts.   Is it inside the FreeCad files or KiCad files somewhere?

jlecomte, 2024-02-09:
OK, I forgot to list that in the BOM. Sorry.

jlecomte, 2024-02-09:
Fixed in [this commit](https://github.com/jlecomte/ascom-oag-focuser/commit/b9ac257cc9738d94fbbac5315695e74b7e7cab98). Thanks!

closed 2024-02-09 by jlecomte