jlecomte/ascom-oag-focuser #10: Notes on Celestron OAG Dimensions

Endymion42, 2024-02-09:
Attached are some notes on the critical dimensions of the Celestron OAG should you decide at some point to make a design for it.

[Notes on Celestron OAG Dimensions for ASCOM-OAG-FOCUSER Picture 1 .pdf](https://github.com/jlecomte/ascom-oag-focuser/files/14216520/Notes.on.Celestron.OAG.Dimensions.for.ASCOM-OAG-FOCUSER.Picture.1.pdf)
[Notes on Celestron OAG Dimensions for ASCOM-OAG-FOCUSER Picture 2 .pdf](https://github.com/jlecomte/ascom-oag-focuser/files/14216521/Notes.on.Celestron.OAG.Dimensions.for.ASCOM-OAG-FOCUSER.Picture.2.pdf)

jlecomte, 2024-02-23:
I will not support the Celestron OAG. I added a section to the README that documents OAG compatibility. Thanks!

closed 2024-02-23 by jlecomte

Endymion42, 2024-02-26:
Here's a completed design for the Celestron OAG.   If acceptable, would you add it to your project for people who want to print and use it themselves?

Regards,
John

[CelestronOAG.zip](https://github.com/jlecomte/ascom-oag-focuser/files/14412291/CelestronOAG.zip)
[image]
[image]
[image]
[image]
[image]

Endymion42, 2024-02-26:
https://github.com/jlecomte/ascom-oag-focuser/assets/92014533/0007f066-bc40-46ed-a3a4-cc8e99c39e2c

Video in action.

jlecomte, 2024-02-27:
This is great! I see you attached an OpenSCAD file. Let me see if I can import it in FreeCAD. Thanks!

Endymion42, 2024-02-27:
OpenSCAD file is a bit messy - this may help:

BaseA is
rotate([180,0,0])
CelestronBaseA();

BaseB is
rotate([180,0,0])
CelestronBaseB();

Cover Plate is
TopCover();

Gear (60 teeth) is
CelestronGear();

Pinion (16 teeth) is
MotorPinion();

jlecomte, 2024-02-27:
I added the files to the GitHub repository, and credited you for this contribution. Thank you!

Endymion42, 2024-02-27:
Notes on printing the Celestron OAG parts.

I used 0.2mm with 15% fill.   No supports used on any parts but I do recommend a brim around BaseB as there is a possibility for the print to mess up at the end of the semicircular clamp if you don't.