jlecomte/ascom-oag-focuser #16: ASCOM Conform and ConformU errors and issues with DarkSkyGeek's OAG Focuser ASCOM driver

dlkeng, 2024-07-02:
Using the ASCOM **Conform** tool with the DarkSkyGeek's OAG Focuser ASCOM driver, the following error occurs:
```
11:06:06.354 SupportedActions   ERROR    Actions must be strings. The type of action 1 1 is: Int32

Your driver had 1 error, 0 warnings and 0 issues 
```

Using the ASCOM **ConformU** tool with the DarkSkyGeek's OAG Focuser ASCOM driver, the following issues occur:
```
11:01:34.326 SupportedActions   ISSUE    Unexpected error: Specified cast is not valid.
11:01:34.329 SupportedActions   ISSUE    Specified cast is not valid.

11:01:40.697 Issue Summary               
11:01:40.699 SupportedActions   ISSUE    Unexpected error: Specified cast is not valid.
11:01:40.701 SupportedActions   ISSUE    Specified cast is not valid.
```

[Conform Log.txt](https://github.com/user-attachments/files/16071560/Conform.Log.txt)
[ConformU Log.txt](https://github.com/user-attachments/files/16071563/ConformU.Log.txt)

jlecomte, 2024-07-02:
Indeed, there is a small bug in the driver code. I'll fix it. Thanks for reporting!

jlecomte, 2024-07-02:
Fixed in release 1.3.1. Thanks!

closed 2024-07-02 by jlecomte