jlecomte/ascom-oag-focuser #17: ASCOM.DarkSkyGeek.FocuserApp.exe unhandled exception

dlkeng, 2024-07-02:
The **_ASCOM.DarkSkyGeek.FocuserApp.exe_** application generates the following unhandled exception when attempting to "**Set Zero position!**" when clicking the either of the left-arrow buttons when the Focuser is moving.

[image]

This does not occur when attempting to click the right-arrow buttons when the Focuser is moving because the right-arrow buttons are disabled while the Focuser is moving. To be consistent, the left-arrow buttons should also be disabled when the Focuser is moving.

dlkeng, 2024-07-02:
Correction, I incorrectly described the operations in the reverse order - in actuality, attempting to "Set Zero position!" when the Focuser is moving generates the unhandled exception. When moving the Focuser with the right-arrow buttons, the "**Set Zero position!**" button is disabled preventing the issue, but when moving the Focuser with the left-arrow buttons, the "**Set Zero position!**" is **NOT** disabled. So, the "**Set Zero position!**" should be disabled when moving with any of the arrow buttons.

jlecomte, 2024-07-02:
Yes, I was able to reproduce. Fix incoming!

closed 2024-07-02 by jlecomte