jlecomte/ascom-oag-focuser #18: Adapter for ZWO OAG-L

naavis, 2024-09-04:
Hi! I'm sorry if this is a dumb question, but could you clarify what the adapter piece is needed for when using the OAG motor focuser with the ZWO OAG-L? I printed the parts and am currently assembling the focuser. My OAG is at the local observatory in active remote imaging use, so I don't have it at hand now. Maybe the need for the adapter becomes clear when I actually go install the focuser parts, but it would be nice to understand ahead of time what the adapter is for.

Judging by the shape and threading, does it somehow go between the OAG and the 1.25" helical focuser?

jlecomte, 2024-09-04:
For ZWO OAG-L, you can either make the custom designed version, in which case you won’t need an adapter, or you can make the standard version and use the adapter to fit the ZWO helical focuser (M42 female thread)  to the ZWO OAG-L body (4 tiny screws)

closed 2024-09-04 by jlecomte

naavis, 2024-09-04:
I am not sure I entirely follow. The OAG-L comes with a helical focuser attached. Why would I need the printed adapter?

jlecomte, 2024-09-04:
Exactly! You don’t need the adapter. The reason I created the adapter is for people who might own multiple different OAGs and they want to swap the focuser between those OAGs. During the crowdfunding campaign, I had several such requests.

naavis, 2024-09-05:
Alright, thanks for the clarification. And great project! Solves exactly the problem I've been having a while with my imaging setup.