jlecomte/ascom-oag-focuser #20: Two instances of ASCOM Focuser Driver

pengsloth, 2024-12-26:
After seeing your video I was intrigued of using this ASCOM driver not only for adjusting focus of an OAG but also the camera itself.

Voyager is an amazing automation software however the developer refuses to add the ability to do filter offsets. Using your driver also an external use of filter offsets that is transparent to Voyager which is wonderful! 

My question is, since I'm using this with the main filter wheel and main camera focuser could I set up a second instance to also operate the OAG focuser.

So the main filter wheel changes from Lum to Ha which changes the main focuser by -140 steps as well as changes the OAG focuser by say 80 steps.

jlecomte, 2024-12-31:
You cannot have 2 instances of the ASCOM focuser driver running at the same time on a PC. You need to use two separate PCs.

closed 2024-12-31 by jlecomte

pengsloth, 2024-12-31:
Would it be possible then to have both the main camera focuser and the OAG focuser available in the Proxy driver? So when a filter is changed it would adjust the main camera focus (technically NINA can do this already but Voyager won't add the ability) and also adjust the focus of the OAG to match the change?

jlecomte, 2024-12-31:
No, it would not be possible. May I recommend you switch to NINA? :)