jlecomte/ascom-oag-focuser #22: Small patch to eat \r character when manually testing the OAG-EAF

michael-ring, 2025-02-23:
When manually testing the OAG Focusser Firmaware on my Mac I saw that the serial command does \r\n at line ends, so it is not possible to test the response to the commands because commands are not detected.

This small patch will fix this:

diff --git a/Arduino_Firmware/Arduino_Firmware.ino b/Arduino_Firmware/Arduino_Firmware.ino
index 2536676..053b05d 100644
--- a/Arduino_Firmware/Arduino_Firmware.ino
+++ b/Arduino_Firmware/Arduino_Firmware.ino
@@ -129,6 +132,7 @@ void setup() {
 void loop() {
     if (Serial.available() > 0) {
         String command = Serial.readStringUntil('\n');
+        command.replace("\r","");
         if (command == COMMAND_PING) {
             handlePing();
         }

jlecomte, 2025-02-23:
Please, open a pull request. That will make it a lot easier to review and merge. Only then can I close this issue. Thanks!

jlecomte, 2025-02-24:
Done in https://github.com/jlecomte/ascom-oag-focuser/pull/24. Thanks!

closed 2025-02-24 by jlecomte